prusa3d/PrusaSlicer #8652: Right click menu when splitting to parts is missing the ability to delete

webRat, 2022-08-08:
**Is your feature request related to a problem? Please describe.**
I'm surprised when splitting to part that the right click menu is missing the basics. The model I'm working with has 20 something parts, and I wanted to select 8-9 of them, and delete them.

**Describe the solution you'd like**
Split-to-parts, select multiple things by holding shift, right click, delete.

**Describe how it would work**
Split-to-parts, select multiple things by holding shift, right click, delete.

**Describe alternatives you've considered**
Someone pointed out that Edit->Delete is available to me in the prusa community. This helps, but ideally, I'd like all the tools I normally have available in the right click menu.

**Additional context**
For a single right click context menu:
[image]

Split to parts, right click:
[image]

neophyl, 2022-08-08:
Why are you worried about the right click options ?  Select the parts you want to delete and then just hit the Del key.  Maybe its not on the right click options as its directly selectable with a key press.

TojikCZ, 2022-08-08:
He wasn't worried, he probably just stuck to what he knew best.
If the slicer can do it, and it's in the menu for single instance, there is most likely no reason for it to be missing for multi instance, it might have been an oversight. If it's an easy fix and can improve the user experience for people, why not?

closed 2026-09-29 by prusaslicer-bot[bot]