prusa3d/PrusaSlicer #8653: Allow user-defined GCode macro definition variables

puterboy, 2022-08-08:
My GCode often repeatedly references a long and composite GCode expression.
It would be great if we could create user-defined macro variables with meaningful names that can be assigned to such expressions.
This would clean up the GCode making it easier-to-read and less likely to omit changes/make mistakes when editing.

For example, in my temperature tower G-code, I repeatedly use the expression:
```int(layer_z-1.4-layer_height+.000001)/10)
```

It would be really great to assign these to a macro definition variable

stepikovo, 2022-09-12:
Hello @puterboy, can you please edit your request, looks like your example is missing.

puterboy, 2022-09-12:
Weird - it's there in the edit but doesn't show up.... Fixed it.

bubnikv, 2023-04-05:
Global and local variables were added to PrusaSlicer 2.6.0-alpha6.

closed 2023-04-05 by bubnikv