prusa3d/PrusaSlicer #8656: Add Option to "Cut" Function to Auto-Generate Alignment Holes

degroof, 2022-08-09:
I was just playing with PrusaSlicer and, when you add a negative volume (e.g. a cylindrical void) to an object, then perform a cut that bisects that volume, the hole is preserved in both cut pieces.

It's not that much of a stretch to add a feature that uses this to create, say 1.8mm diameter holes at the cut plane while performing a cut. The user could position the holes manually, or the slicer could automatically distribute them over the interface area (J holes per K mm², plus one hole for any isolated area > L mm²). Then, when you print the two pieces, you end up with alignment holes sized to accept a short length of filament, keeping everything nicely aligned for gluing.

I mean, it's possible to do this manually with the current slicer, but it'd be great if it could be automated.

[image]

[image]

[image]

dartrax, 2022-08-09:
I really like that Idea, but we have this already in #803 and #5553 and #2357 😉

lukasmatena, 2023-02-01:
Implemented in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena