prusa3d/PrusaSlicer #8658: Insert comment in gcode before new object is printed in sequential printing

Hologos, 2022-08-09:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Since color change is not supported in PrusaSlicer, it would great if you insert a comment announcing new object in sequential printing so I can manually add my color change sequence.

**Describe how it would work**
Emit comment in the gcode when new object is being printed in sequential printing.

**Describe alternatives you've considered**
None

**Additional context**
None

neophyl, 2022-08-09:
You could enable the option to label objects.  While not exactly what you want it add in the object for which the gcode is for, for example -
; printing object Cord Plug back   
You could then just search for the first instance used of the name of the object you need

Hologos, 2022-08-09:
I am such a noob, there is the Between Object G-Code in custom g-code section. *facepalm*

Thanks @neophyl

closed 2022-08-09 by Hologos