prusa3d/PrusaSlicer #8662: Variable Layer Height to minimize overhang and support using topographic algorithm

physicsimulation, 2022-08-09:
I am used to Cura's Adaptative Layer Height that is perfect for my topographic print out, and it is really intuitive
 - Cura's Maximum Variation have similar consequences as Prusa's Printer Settings-Extruder-Layer Height Limits
     = Kind of "Actual layer height within Quality Layer Height +/- Maximum Variation"
 - Cura's Variation Step Size is the maximum "delta layer height"
     = Kind of "Next layer height = Actual layer height +/- variation step size function of Cura's Topographic Size"
 - Cura's Topographic Size is adjusted to fit average layer height. FOR EXAMPLE, with Adaptative Layer Height disabled and the Quality Layer Height set to 0.20, the slicer give 250 layers for 50 mm height object.
     = with Adaptative Layer Height ENABLED, and a well choose Topographic Size, you could get near 250 layers for a similar print time,
     = The advantage of the Cura's method is that you got less overhang then no adaptative layer height.
May be it is less accurate then Prusa's Wasserfall algorithm but intuitive. (https://help.prusa3d.com/article/variable-layer-height-function_1750?fbclid=IwAR16QaHDY-Dmhk0acZKMQHOtRde87bP_ph_-xDrPVmQPLzoKZmV32hDul1A)
You could try to slice a topographic object (i.e. https://touchterrain.geol.iastate.edu/) with Cura's adaptative layer enabled to appreciate overhang reduction.

I would appreciate to have, in PrusaSlicer's Variable Layer Height interface, an option to select the algorithm "Wasserfall/Topology" with new parameters for Topographic.

Topographic parameters could be "Variation Step Size" and "Targeted Average Layer Height", still offering "Smooth" and your wonderful manual editing. 
[image]

kubispe1, 2022-08-15:
Thank for your recommendation. We will observe it and add it to backlog of out feature requests.

murk-sy, 2022-10-04:
+1, for a lot of cases using curves, just a maximum step (until minimum layer height is reached) would already improve ease of use - i.e. think something like a chess pawn.

[image]

Of course, doing a smooth layer height transition would be a *bit* more work than just calculating max step over.

closed 2026-09-29 by prusaslicer-bot[bot]