prusa3d/PrusaSlicer #8664: Detect thin walls now working in  2.5.0-alpha3

dan90ned, 2022-08-10:
### Description of the bug

Detect thin walls now working in  2.5.0-alpha3. It is not possible to check the box 
[image]

### Project file & How to reproduce

Go to Print Settings -> Layers and Perimeters -> check box of Detect thin walls

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

 2.5.0-alpha3

### Operating system

Windows 10 x64

### Printer model

MK3S+

neophyl, 2022-08-10:
Detect thin walls is greyed out and not available with the Arachne perimeter generator option (which is now the default on 2.5.x).  However it looks like your colour scheme obscures the fact that the tickbox is greyed out.  If you change to the normal mode instead of the experimental dark mode you will be able to see that.

So the 'bug' in this case is that the experimental dark mode is not able to show when a tickbox has been greyed out and not available.

seanmcne, 2022-08-12:
@dan90ned it might be a good idea to update the title of the issue to something like: 
UI Bug Dark Mode on Windows: Detect thin walls checkbox disabled status incorrect on 2.5.0-alpha3

kubispe1, 2022-09-14:
Hello guys. I think, I find new reported same issue, but it is described in little better way, so I will let open newer one... But thanks.
duplicate #8877

closed 2022-09-14 by kubispe1

RoostOne, 2022-12-20:
Hi, maybe there co