prusa3d/PrusaSlicer #8665: Slice Defect

chris17453, 2022-08-10:
### Description of the bug

A model when sliced is dissimilar to non sliced model. Specifically a several millimeter gap in the object appears and create a split in the object which is otherwise solid. This does not happen when the same file is loaded into the original "Slic3r" or "IdeaMaker".

### Project file & How to reproduce

Open the STL model
Slice the model with no supports
View the slice. the bottom portion a very large gap is created creating almost 2 different objects.

[model.zip](https://github.com/prusa3d/PrusaSlicer/files/9299652/model.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.2+linux-x64-GTK3-202204251120

### Operating system

Fedora 36 -> Linux 5.18.15-200.fc36.x86_64 x86_64

### Printer model

PRUSA MK3S+

neophyl, 2022-08-10:
Thats because your model has errors.  When loaded it displays the yellow triangle in the tree list indicating that it has problems.  
Fix it and it slices correctly.

[016_30_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/9300130/016_30_fixed.zip)

chris17453, 2022-08-10:
Thanks for the fix. It's not my object. What was the process to fix it? I'm just wondering if I missed a setting or process with the software that i was unaware of. Mostly because this doesn't happen in the other slicer applications. Maybe I missed a setting?

Thanks so much

chris17453, 2022-08-10:
Thank you for your help, literally has been invaluable!

closed 2022-08-10 by chris17453