prusa3d/PrusaSlicer #8670: PrusaSlicer Programmabsturz bei Auswahl Variable Schichthöhe

Panther5241, 2022-08-11:
### Description of the bug

Beim klicken auf die Schaltfläche Variable Schichthöhe wird PrusaSlicer beendet.
Systeminformationen und Screenshot im Anhang
[Systeminfo und Fehlerbild.pdf](https://github.com/prusa3d/PrusaSlicer/files/9307297/Systeminfo.und.Fehlerbild.pdf)
.

### Project file & How to reproduce

[Stuhlgleiter.zip](https://github.com/prusa3d/PrusaSlicer/files/9307307/Stuhlgleiter.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.2+linux-x64-GTK2-202204251127

### Operating system

Linux Mint 20.3 Cinnamon 64-bit

### Printer model

Prusa Mini+

kubispe1, 2022-08-11:
We tried your .3mf in Win 10 Pro and Ubuntu. We did not find any issue. Maybe someone else with Linux Mint could confirmed it ? ....

Panther5241, 2022-08-11:
Ich habe gerade mein Linix Mint aktualisiert und die PrusaSlicer 2.5.0-alpha3 getestet.
Problem besteht weiterhin und auch mit der alpha3.
Ich versuche jemanden zu finden, der es auf seinem Linux Mint versucht.

I just updated my Linix Mint and tested the PrusaSlicer 2.5.0-alpha3.
Problem persists and also with the alpha3.
I am trying to find someone to try it on their Linux Mint.

closed 2026-09-29 by prusaslicer-bot[bot]