prusa3d/PrusaSlicer #8671: M107 called twice during startup

gudvinr, 2022-08-11:
### Description of the bug

During startup routine M107 called twice:
* after standard accel/feedrate setup before custom g-code
* after custom filament gcode

I think second M107 is redundant and in fact, messing with custom g-code which might or might not set fans for some reason which will be disabled.

### Project file & How to reproduce

1. Slice file.

[xyzCalibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/9308063/xyzCalibration_cube.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Arch Linux

### Printer model

Artillery Genius Pro

kubispe1, 2022-08-15:
It is theoretical issue that could in some situations cause problem. But you are totally right. I add enhancement label.

gudvinr, 2022-08-22:
So, I found out how that happens.

First, `set_fan(0)` explicitly called in startup code:
https://github.com/prusa3d/PrusaSlicer/blob/b62689437ab6fcec8ae940a9664eaf5c83620fec/src/libslic3r/GCode.cpp#L1243

Then, inside `process_layers` code that checks whether or not you should ramp up the fan sets `M107` because object initialized with `m_fan_speed = -1`:
https://github.com/prusa3d/PrusaSlicer/blob/b62689437ab6fcec8ae940a9664eaf5c83620fec/src/libslic3r/GCode/CoolingBuffer.cpp#L737

kubispe1, 2022-08-23:
Thanks for extra info. Hope there will be time to investigate it more.

closed 2026-09-29 by prusaslicer-bot[bot]