prusa3d/PrusaSlicer #8674: Feature request: Integrate with Printables / Thingyverse

ondrejch, 2022-08-12:
Ii would be great to select repository model directly in the slicer, and add to buildplate from the slicer. 

Cura has Thingyverse Marketplace integration. It does not allow going into one's collections, but it works.

gudvinr, 2022-08-12:
Simpler and more reasonable way is to use custom URI schema on printables and add "open in slicer" button which will open model in slicer of your choice.

For example for this page: https://www.printables.com/model/246961-cascade-card-wallet-print-in-place add button with link `printables://prints/246961/packs/1249889_0fdc7194-c9d3-4efe-bad6-a8ace6eef8c6/cascade-card-wallet-print-in-place-model_files.zip` (which leads to archive with all files, with similar links for individual stls).  
_Links are just examples, in is not suggested format nor schema to use_.

With that approach _any_ slicer can register itself as URI handler for this link.  
Also there is no need to integrate model browser in slicer which either needs browser component to be included or some sort of API on server which adds a lot of complexity.

lukasmatena, 2023-02-01:
Implemented in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena