prusa3d/PrusaSlicer #8675: Cannot Compile On MacOS 13 Beta

lbibass, 2022-08-12:
### Description of the bug

after compiling the dependencies, attempting to generate the main project file fails. 

This is the command I used to generate everything. This worked in the past. 

`cmake .. -G Ninja  -DCMAKE_PREFIX_PATH=../deps/build/destdir/usr/local/lib/ -G Ninja -DCMAKE_BUILD_TYPE=Release -DSLIC3R_STATIC=1`

```-- Could NOT find OpenVDB (missing: OpenVDB_INCLUDE_DIR OpenVDB_LIB_COMPONENTS openvdb) (Required is at least version "5.0")
CMake Warning at cmake/modules/OpenVDBUtils.cmake:129 (message):
  vdb_print not found! OPENVDB_PRINT-NOTFOUND
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:317 (OPENVDB_ABI_VERSION_FROM_PRINT)
  CMakeLists.txt:526 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:324 (message):
  Unable to determine OpenVDB ABI version from OpenVDB installation.  The
  library major version "" will be inferred.  If this is not correct, use
  add_definitions(-DOPENVDB_ABI_VERSION_NUMBER=N)
Call Stack (most recent call first):
  CMakeLists.txt:526 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:345 (message):
  IlmBase::Half can not be found!
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:367 (just_fail)
  CMakeLists.txt:526 (find_package)

CMake Error at CMakeLists.txt:531 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.```

### Project file & How to reproduce

N/A, bc this is just compiling. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Latest Git

### Operating system

MacOS 13.0 Beta 

### Printer model

Annex K3

lukasmatena, 2022-08-15:
It would help if you stated which branch are you trying to build and how you built the dependencies in `deps/build` (especially from which branch/commit). You might be trying to build `master_250` with deps built from the `master` branch. If so, just rebuild the deps.

lbibass, 2022-08-15:
I was using the deps compiled from the master_250 branch. 

For whatever reason, it does not seem to create the files needed for compilation in the dependencies folder.

lukasmatena, 2022-08-16:
@lbibass It should work then. Did you clean the build folder completely in both cases (deps and PS)? Weird errors often appear when using cached CMake configuration.

lbibass, 2022-08-16:
> @lbibass It should work then. Did you clean the build folder completely in both cases (deps and PS)? Weird errors often appear when using cached CMake configuration.

I have done a completely clean clone from the github, and immediately switched to the master_250 branch. after compiling the deps, and attempting to compile the main build folder, it still fails.

lbibass, 2022-08-24:
Any updates on any testing I can do to make this fix happen sooner?

lukasmatena, 2023-02-09:
Nobody else has this issue, apparently not a problem in CMake files. The only thing that comes to mind is that it might help to use absolute path for `CMAKE_PREFIX_PATH`. Either way, we do not have resources to debug this further. Closing.

closed 2023-02-09 by lukasmatena