prusa3d/PrusaSlicer #8677: Merge functionality for profile comparison

Ic33s, 2022-08-13:
**Is your feature request related to a problem? Please describe.**
1. It's a pain to keep main profiles and their variations consistent where wanted, when changes are done.
2. It's also a pain to get material profiles consitent over different printers (which is important because I need to create different printer profiles for different nozzles)
3. Get consistent setting changes accross different printer profiles (e.g. for different nozzle sizes)

**Describe the solution you'd like** - **Describe how it would work**
Current workflow:
=> Opening Compare ->Look on differences -> Closing Compare -> Switch profile to the one I want to change -> Apply change => Start again.

Wanted workflow:
=> Opening Compare -> Look on differences -> Apply manual changes directly in this window or at least take over changes from the compared profile. This should be able for user selected changes of course not an "Take over all changes")

**Describe alternatives you've considered**
Alternative or better extended final solution :); "Improved profile handling"
An improved profile handling where I inherit changes from an other profile (even accross printer profiles). But this seems much more of an effort because there is much more needed to handle cases where I want to change something in a child profile (=> decision: Do I want to overwrite it in the child profile only or in the current origin) or to go back to the inherited value (but I think this works allready)

Hope for feedback. 

Thank you very much.
Best Icees

gudvinr, 2022-08-14:
I believe it is duplicate of #5384

rtyr, 2022-08-14:
Closing as a duplicate.

closed 2022-08-14 by rtyr