prusa3d/PrusaSlicer #8678: Build plate black after latest AMD adrenalin drivers installed for AMD graphics cards

urshurak776, 2022-08-13:
### Description of the bug

Build plate black after latest AMD adrenalin drivers installed for AMD graphics cards.  Thread on Prusa Forums about it.  Several people have the problem.

### Project file & How to reproduce

Use a PC with an AMD graphics card.  Load latest AMD Adrenalin drivers

[image]
.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10 64bit

### Printer model

Prusa Mini and MK3S - issue happens with either printer selected

rtyr, 2022-08-15:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8508, https://github.com/prusa3d/PrusaSlicer/issues/8603, https://github.com/prusa3d/PrusaSlicer/issues/8417.

See comments there for a possible solution. I am closing this as a duplicate.

closed 2022-08-15 by rtyr

hejllukas, 2022-08-19:
We tried to investigate this issue, and it seems that the two latest AMD GPU drivers, 22.8.1 and 22.7.1, probably contain some bug that is causing missing bed texture. For some reason, those new drivers have an issue when mipmaps are generated manually on CPU sides combined with texture compression.

We generate mipmaps manually on the CPU side because we previously had issues on some GPUs with generating mipmaps on the GPU side.
And for some reason, those new drivers stopped accepting texture mipmaps generated on the CPU side.

So we made a workaround that for AMD GPUs with drivers newer than 22.6.1 (the latest working driver), we generate mipmaps on the GPU side, which seems to work correctly now.
This workaround will be part of the upcoming PrusaSlicer 2.5.0-rc1, so the texture of the bed should start showing again.

For older PrusaSlicer versions, the only way (at the moment) to get texture showing is to downgrade the AMD driver to 22.6.1 or 22.5.1 (WHQL).

urshurak776, 2022-08-19:
Excellent!  Thank you.

bubnikv, 2022-08-19:
Please try with PrusaSlicer 2.5.0-beta1. The AMD GPU driver issue should be resolved.

urshurak776, 2022-08-19:
Worked perfectly!!!

Thank you.