prusa3d/PrusaSlicer #8679: Preview gets into invalid state after Print Settings changed.

vladimir-aubrecht, 2022-08-13:
### Description of the bug

When user tries to change any settings in Print Settings after slicing, than slicing Preview will start showing invalid data (for example model disappear, but supports stays, etc.) if any (Preview is just empty).

See screenshots (I just changed "extra perimeters if needed", but it does it probably with any settings):
Preview:
[image]

3D editor view:
[image]

There is simple workaround: switch back to 3D editor view and back to preview, thou it's still causing frustration as it leads to additional steps.

This bug exists at least since version 2.3.

### Project file & How to reproduce

1) Open model in PrusaSlicer.
2) Go to Preview view.
3) Go to Print Settings tab and change some values.
4) Go back to Plater tab.
5) Preview view will be broken now.

Same issue is occuring when Print Settings is changed from the sidebar (by for example picking different layer height).

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.0+win64

### Operating system

Windows 10 Pro

### Printer model

Prusa Mk3S+ with MMU2S

neophyl, 2022-08-14:
When you change a setting that effects the model then PS needs to reslice the model to be able to preview the layers.  Notice the button in the bottom right says Slice Now.  If you click it it will re-slice and show you the new preview.  Depending on the settings you alter, depends on if anything gets 'left' behind in the preview.  In your example you changed a setting that effects perimeters so the support isnt effected.  If you change infill % then the perimeters would be left but the infill disappears as the preview is no longer valid for the changed settings, until its recomputed. 

When you change back to the 3d window and then click back to the preview thats triggering the slice too as that's what its supposed to do.  You could also enable Background processing in preferences>General if you want.

So its not an issue but the result of your configured settings.  Many people do not like to use background processing and prefer the manual slice option.

vladimir-aubrecht, 2022-08-15:
Oh thank you, didn't notice the settings! 😊 

I think behaviour itself make sense, thou I think it might be confusing for people coming to Prusa Slicer. What I see as a problem is missing visible indication of what happened. For a user it looks like suddenly something random disappeared...

One of the solutions I can think off would be to keep what was rendered originally, but part which is invalid could be faded. Or maybe just introduce some notification like: "Due to change of settings, rendering of XY was removed. Click here to re-slice."

kubispe1, 2022-08-16:
At the moment slicer behaves in way that neophyl clearly described. Could be background processing suitable for you? If you expect another slicer response write it down and I will label it as  feature request and add it to out backlog.

vladimir-aubrecht, 2022-08-16:
Yes, it is, thank you for pointing that out 😊 Sorry for confusion, my last comment is indeed just suggestion how to improve UX more, rather than than advocating existence of bug (as @neophyl clearly explained, it's behaving by design).

kubispe1, 2022-08-19:
That’s ok and thanks for your oppinion. In case we will register more similar UI requirements, we wil try to improve it. At the moment I am closing this issue.

closed 2022-08-19 by kubispe1