prusa3d/PrusaSlicer #8680: Volumetric flow calculations

bitingsock, 2022-08-14:
### Description of the bug

In slice view I can see that speed, line width, and layer height stay the same but volumetric flow changes. How is this possible?
[image]
[image]
[image]
[image]

### Project file & How to reproduce

Set print speeds fast enough that everything would be limited by set volumetric limit.
[simple_storage_box_lid.zip](https://github.com/prusa3d/PrusaSlicer/files/9333302/simple_storage_box_lid.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha3+win64-202207151230

### Operating system

Win10pro

### Printer model

Ender 3 pro

rtyr, 2022-08-15:
I was able to reproduce it with 2.5.0-alpha3, but not with current master, so it should be fixed. Please retest with upcoming 2.5.0-beta1.

closed 2022-08-15 by rtyr

bitingsock, 2022-08-15:
was it just a graphical bug or what?