prusa3d/PrusaSlicer #8682: Stringing as a result of not retracting sometimes before travel!?

Doprintityourself, 2022-08-15:
### Description of the bug

Hello I have a problem wiht strange stringing now and then on my modles 
I have checked with 2.4.2 and 2.5.0a3
[image]
printed with "only retract when crossing..." on and off, no difference.

It looks to me that the slicer does forget to retract sometimes and I am not good enough to hunt this down in the code

Does anybody have same problems, is it know that the last Prusa Slicer versions have a bug in this direction?
Thank you for answering
[Retracts missing during print.zip](https://github.com/prusa3d/PrusaSlicer/files/9335815/Retracts.missing.during.print.zip)

### Project file & How to reproduce

At least try to find out from the visual check where travel without retraction occures and try to find the error in the gcode.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2 2.5.0a3

### Operating system

Windows10

### Printer model

TwoTrees SaphireS

rtyr, 2022-08-15:
I printed your Gcode without problem issues (considering it was sliced for different printer), so the g-code output is OK. The issue on your photo looks more like extrusion, not just stringing (it also doesn't correspond with travel paths). Maybe your hotend/extruder cannot handle such volumetric speed and some extrusions are dragged in the air. You may try to print it with half-speed for example.

[image]

Doprintityourself, 2022-08-15:
Hm, interesting thank you very much for verifying live.

the volumetric speed is about 6.5 max in that areas
[image]

and the system is equipped with a E3D Volcano 0.6 Nozzle that sould do 10+ easily

Did you print it with PETG as well?
Verified again and going throug each travel on the 3dp Logo found at least there always a rectract before travel.
[image]
Retract itself is alreade high with 5mm, so not shure what this massive extrution througout the travel is coming from

neophyl, 2022-08-15:
Its not extrusion throughout travel, its the extrusion of the perimeter that should be stuck to the layer below that isnt sticking and so is being dragged across the inner circle till it does actually stick further along.  That as rtyr pointed out is an indication of a physical problem of some sort with your extruder.

Doprintityourself, 2022-08-15:
hm, I cannot agree totally,
as I followed  some of the arranged strings in the picture above.
[image]

it seems I can directly identify the strings being produced during travel 
same on the other side if you like to compare to the printed out first picture
[image]

and the outer perimeter do not have gaps If they should not have been printed correctly they should be visible at the the part
but I cannot see gaps in the skin!?

stepikovo, 2022-08-23:
Your issue really seems to be caused by some hardware problems, it would be better to solve it with Prusa Support and not on GitHub so we can close it.

closed 2024-02-19 by lukasmatena