prusa3d/PrusaSlicer #8686: seam - still ugly

yahbluez, 2022-08-15:
seam needs to get a kind of gravity mode.

issue

On a curved face, seam runs down in an angel (or like a drunken angel).

solution

let it be on the same X or Y coordinate and it will be straight down like gravity does.

kubispe1, 2022-08-16:
Would be more appropriate 'rear' option of seams for you and your case? 
Can you provide more description of your issue (screenshot, 3mf and so on) ?

yahbluez, 2022-08-21:
No that did the same on the other side - no difference in quality.

The point is, an aligned seam, no matter of in front or rear, should be a straight line down to the ground, and not a path of a drunken snail.

Godrak, 2022-08-23:
Can you provide more description of your issue (screenshot and 3mf) ?

Witawat, 2022-09-13:
https://github.com/prusa3d/PrusaSlicer/issues/8841

kubispe1, 2022-09-15:
This issue is duplicate. And also there is no response for a long time. So, I close it. 
Thanks

closed 2022-09-15 by kubispe1