prusa3d/PrusaSlicer #8688: .obj file generates error

b-jacoby, 2022-08-16:
### Description of the bug

Attempt to load .obj file generates generic "Prusaslicer has encountered an error.  Loading of a model file has failed".

### Project file & How to reproduce

File -> Import -> Import STL/OBJ/AMF/3MF, etc...

or 

drag and drop onto the empty workspace.

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-beta1+win64

### Operating system

Windows 11

### Printer model

Prusa Mini (4.3.4 bootloader)

kubispe1, 2022-08-16:
Can I ask you to provide us .obj file that makes you trouble.

kubispe1, 2022-09-01:
This conversation is inactive. So, I am closing it. Please in case try some of our new releases and let us know, if troubles persists.

closed 2022-09-01 by kubispe1