prusa3d/PrusaSlicer #8690: Setting 100% infill on an object crashes prusaslicer

abudden, 2022-08-16:
### Description of the bug

I opened an STL, flipped it over and tried to set the infill density for that object only.  It popped up a message saying that gyroid didn't support 100% infill and then when I clicked okay, prusaslicer crashed.

I would expect it to either (a) add a custom infill pattern to the object automatically, (b) refuse to slice until the issue is fixed or (c) reduce the infill density automatically (in rough order of preference).  I wouldn't expect it to crash.

The obvious workaround is to add the infill pattern as a custom setting as well (in which case it works as expected), but I thought it was worth mentioning this minor bug.

### Project file & How to reproduce

[Makita_RT0700_Dust_Collection_Lid.zip](https://github.com/prusa3d/PrusaSlicer/files/9351207/Makita_RT0700_Dust_Collection_Lid.zip)

Open the attached STL and flip it over so the top face is on the bed:

[image]

Right-click, click Add Settings and Infill:

[image]

Select infill density only:

[image]

Type 100 in the infill box:

[image]

Click okay and watch **PrusaSlicer crash**.

If you Right-click and click "Infill" in the top-level menu, it adds "Fill density" and "Fill pattern" as settings and then it works as expected.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Windows 10

### Printer model

Prusa i3 Mk3S+

kubispe1, 2022-08-16:
Sorry, but I am not able to reproduce your issue. I tried versions 2.4.2+win64 (same as you), 2.5.0-beta1 + win64, and even v 2.4.2 for Ubuntu. In all situations PS switches infill to rectilinear without crash. 
So, I would ask you to attach .3mf before you add there Fill density. I will check it again.

abudden, 2022-08-16:
That's weird... I just tried again and saved the .3mf after adding the fill density option and before changing the density to 100%.  I then changed the density to 100% and it added the fill pattern and set it to rectilinear.

I don't understand what's different: when I was writing the issue I tried it at least 8 times (to make sure the steps were correct and consistent) and it crashed every single time.  Now I can't reproduce it either...

kubispe1, 2022-08-17:
That's strange, but don't worry. If you have a problem or issue again let us know. Thanks

closed 2022-08-17 by kubispe1