prusa3d/PrusaSlicer #8693: GUI: Layout of "Object Manipulation"-Panel is broken (Linux GTK3)

deepthought-64, 2022-08-17:
### Description of the bug

Using Linux and GTK3 I see the layout of the components in the  "Object Manipulation" Panel (bottom right corner) in the "Plater" Tab as broken. The labels for `Position`, `Rotate`, etc do not align with the TextBoxes.

See:
[image]).

I am using `Version 2.5.0-beta1+linux-x64-GTK3`

### Project file & How to reproduce

Open PrusaSlicer and look at the `Object Manipulation` panel. No further action needed.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-beta1+linux-x64-GTK3

### Operating system

Arch Linux (5.19.1-arch2-1)

### Printer model

i3 mk3

lukasmatena, 2022-08-17:
I can confirm this issue on my Lubuntu 20.04 with all PrusaSlicer versions back to at least 2.3.3. I don't know if it was always like this or if it broke with some gtk3 system package update.

noisufnoc, 2023-01-13:
Same issue on my Ubuntu Budgie 22.10 machine, running PrusaSlicer-2.5.0

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena