prusa3d/PrusaSlicer #8695: [2.5.0-beta1] Arachne not respecting seam painting

redlukas, 2022-08-17:
### Description of the bug

When using seam painting and seam blocking
[image]

the classic perimeter generator engine respects those blocks/enforcers
[image]

while arachne does not
[image]

My expectation would be that arachne rexpects the painted-on seams just like the classic engine does.

### Project file & How to reproduce

[seam.zip](https://github.com/prusa3d/PrusaSlicer/files/9357607/seam.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-beta1+win64

### Operating system

Windows 10 Pro 21H2

### Printer model

MK3S+ with MMU2S

redlukas, 2022-08-17:
This issue seems to only occur when a wall only consists of perimeters. as soon as the wall is thick enough to contain infill (in the uploaded model i switched from the 0.8mm nozzle to the 0.4mm one) the support enforcers are respected by arachne.
[image]

Mithrandil, 2022-09-05:
THis issue is somewhat still present in 2.5.0-rc2.
The seam (gray) seems to be placed correcly, but not the retraction/deretraction (purple and blue), causing holes on single perimeter walls intersections.

Classic:
[image]

Arachne:
[image]

DragonFinlay, 2022-09-15:
I cannot reproduce this anymore in 2.5.0 final, even after trying Classic/Arachne, different nozzle sizes and seam painting. It now always gives the expected result with the seam being a nice straight line in the middle.

@Author Can you try it yourself and let me know if it still happens for you in the stable version of 2.5.0?

redlukas, 2022-09-17:
yes, tested it with PrusaSlicer-2.5.0+win64-202209060714 and the seam is now aligned both
with classic
[image]

and arachne
[image]

thanks for the fix!

closed 2022-09-17 by redlukas

xGERAx, 2022-09-18:
I still have the seam working incorrectly in the PrusaSlicer-2.5.0+win64 version
[image]
[image]

redlukas, 2022-09-18:
hmm how does that work now? can I transfer ownership of the issue to @xGERAx ? or do I reopen the issue and watch for their confirmation to close it evantually?

xGERAx, 2022-09-18:
I am new here. I don't quite understand how it works)

redlukas, 2022-09-19:
@xGERAx maybe its best if you open your own issue, as it pertains to another version of prusaslicer than is mentioned in my original description

DragonFinlay, 2022-09-20:
We have plenty of open issues about seams currently and we're working on them.
On the issues page, add "label:seam" to the search bar parameters and you'll be able to find them all. :)

[image]

Please find the one that is closest to your case and comment there. Don't forget to include the 3MF.