prusa3d/PrusaSlicer #8696: Top solid infill in vase mode

espr14, 2022-08-17:
**Is your feature request related to a problem? Please describe.**
The top most bottom solid layer does not look good in vase mode.

**Describe the solution you'd like**
Treat the top most bottom solid infill as a top one. This means:
1. enable ironing on the last solid layer only
2. use top solid infill speed
3. use top solid infill extrusion width

It does not mean to create top layers at the top of the model.

**Describe how it would work**
Bottom solid infill number/height sets where solid infill ends in vase mode. The last layer (or more last layers) should be treated as top layer(s). This would make the top layer nicer/smoother.

**Describe alternatives you've considered**
- adding ironing to all solid layers slows the print a lot
- avoiding vase mode makes seems

**Additional context**
We can use number of top solid layers and even minimum thickness (in mm) in vase mode. User could set how many layers will be treated as top layers. Just (unlike non-vase mode) the number (thickness) of top layers must be equal or lower than number (thickness) of bottom layers.
Minimum bottom shell thickness should be enabled also but this was, probably, mentioned in other feature request already.

stepikovo, 2022-08-24:
Can you please provide pictures showing the issue and also the 3MF file with your project?

espr14, 2022-08-24:
[Shape-Cylinder.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9418827/Shape-Cylinder.3mf.zip)
In PS 2.5 RC1 the top solid layer extrusion width a speed works just fine. Just the missing ironing remains.

Poikilos, 2025-11-06:
I would like this as well. The last solid layer is not filled unless I do "All solid surfaces", but I would like it to be filled when I do "All top surfaces" so I don't have to wait for ironing on every layer before the last soild (non-vase) layer.

PrusaSlicer 2.9.3 flatpak

[image]

closed 2026-09-29 by prusaslicer-bot[bot]