prusa3d/PrusaSlicer #8698: command line +data parameter fail

henrikkoch, 2022-08-17:
### Description of the bug

If the path added to the command line is not accessable (server is down ie.) then Slicer just shuts down instead of informing the user that the path for the data dir is not accessable

### Project file & How to reproduce

try to write a wrong path to the data dir

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0 Alpha 2 (but previous versions also fails here)

### Operating system

Win10

### Printer model

Prusa i3MK3+

henrikkoch, 2022-08-18:
Maybe it should be less destructive and just either ignore the Data dir parameter or most best inform the user that the path can't be found.

kubispe1, 2022-08-24:
Hi, I am not sure if I understand of your issue. I step in this way (PS 2.5.0 Alpha 2, WIN10):
1) Open cmd
2) put there PS.exe --datadir NON-EXISTING PATH 
* PS.exe - with correct absolute path
* NON-EXISTING PATH - for example on D:\ that does not exist.
3) Press Enter
And after I get this warnings 
[image]
and this (in CZ)
[image]
4) After PS shuts down.

In case, can you describe your issue in deeper detail. I am not sure, if I reproduce it according your note.

henrikkoch, 2022-08-24:
>    1. Press Enter
>    2. After PS shuts down.

In my case i'm using this:

"C:\Users\henrik\Downloads\PrusaSlicer-2.5.0-rc1+win64-202208191450\prusa-slicer.exe
--datadir X:\Slic3er-settings"

[image: image.png]

But as this data dir points to my NAS which due to power savings were off.
Then slicer broke down as it couldn't find the path

/Henrik

henrikkoch, 2022-08-25:
In my case i'm using this:

"C:\Users\henrik\Downloads\PrusaSlicer-2.5.0-rc1+win64-202208191450\prusa-slicer.exe --datadir X:\Slic3er-settings"

 
[image]

But as this data dir points to my NAS which due to power savings were off. Then slicer broke down as it couldn't find the path

/Henrik

kubispe1, 2022-08-25:
That is totally strange cause in my case it also emits same series of warnings, I showed you before (I followed your last comment). So, I can only guess, that it could be cause, that you have this server path saved somewhere in your system. Have you also tried some random location / path ?

closed 2026-09-29 by prusaslicer-bot[bot]