prusa3d/PrusaSlicer #8701: Filament Shrinkage

ruredi87, 2022-08-17:
Hi. I am printing mostly engineering parts to fit each other. You know different filaments has different shrink rate, so We should scale model by its shrink rate. In prusa slicer we should do it manually for each slice. If I forget to scale it some times  I had  to reprint object. There is horizontional X/Y compensation but it ıs not solving issue because shrink rate should be calculated with percent, not exact number value.

So shortly There should be shrink rate value for store in filament profile and It should scale it automaticly each time. Thanks.

I am uploading sample image from superslicer.
[image]

neophyl, 2022-08-17:
Duplicate issue.  Please see https://github.com/prusa3d/PrusaSlicer/issues/3059

closed 2022-08-17 by rtyr

MrBlackmidi, 2024-02-08:
Hi, is there an update on about implementing this, also z shrinkage aswell?