prusa3d/PrusaSlicer #8703: STEP file fails to load with "cannot load OCCTWrapper.dll" on rpi Bullseye (Linux11)

ericd1949, 2022-08-17:
### Description of the bug

Loaded the arm71 AppImage

The error is given after trying to import a step file:
PrusaSlicer has encountered an error:
cannot load OCCTWrapper.dll

The pi is a pi4

FYI This does not happen on the DavidK AppImage of the same release

### Project file & How to reproduce

[Raspberry Pi 3B mount-1.step.zip](https://github.com/prusa3d/PrusaSlicer/files/9364081/Raspberry.Pi.3B.mount-1.step.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-beta1

### Operating system

Linux 11 (Bullseye)

### Printer model

Mk3S

lukasmatena, 2022-08-17:
Thanks. The dynamic library is missing in the RPi archive for some reason. We have to fix it on the buildserver.
The error message also incorrectly refers to a `.dll` library, although it really tried to open `.so`. I just fixed this in e9f0f4bd0dfb5cadb5fd6db7aa5ca3e7b89a48e3.

ericd1949, 2022-08-17:
Great. Happy to help.

lukasmatena, 2022-08-18:
@ericd1949 Could you please download and test this build?
[PrusaSlicer-2.5.0-beta1+2-linux-armv7l-GTK2-ge9f0f4bd0](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.5.0-beta1+2-linux-armv7l-GTK2-ge9f0f4bd0-202208180611.AppImage)
Thanks.

user: prusaslicer
password: slicer

ericd1949, 2022-08-19:
Lovely, that fixed it. Thanks a lot - I'll continue having fun with it :-)
Sorry to be slow on the uptake, my notifications weren't appropriate.

lukasmatena, 2022-08-19:
Thanks for the confirmation. Closing the issue.

closed 2022-08-19 by lukasmatena