prusa3d/PrusaSlicer #8706: Arachne mode causes major undersized holes

PDBeal, 2022-08-17:
### Description of the bug

When slicing with Arachne with PrusaSlicer-2.5.0-alpha3 on MacOS 12.5.0, holes are way undersized.  Slicing in Classic mode doesn't seem to have this issue.  It's not related to a single layer either, the top and bottom layers and everything in-between is way to small with Arachne.

Testing printing a rectangle with a 1mmx3mm hole for a tie wrap in a part that was 2mm thick.  The tie wrap would pass through the hole when sliced with Classic, but would not go through without cutting additional layers away inside the hole when printed with Arachne slicing.

I will try with the new beta version, but I searched the issues and didn't see anything similar, so I'm not expecting any difference.

[print_test.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/9366508/print_test.stl.zip)

### Project file & How to reproduce

[print_test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9366518/print_test.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0-alpha3+arm64

### Operating system

MacOS 12.5.0

### Printer model

Creality Ender-5 with Duet 2 Maestro, E3D Titan, and E3D Revo Six

kubispe1, 2022-08-18:
Hi, I reached the best results with Arachne + External perimeters first (Layer and parameters/External perimeters first). I think this function is there for such a cases.  I am not sure if you can reach absolute precision for small holes 3 x 1 mm.  It also depends on nozzle size and printer calibration. This is only my opinion.

asteppke, 2022-09-10:
It seems to me that it is not only holes that are undersized but circular features. 

My test case: a 5mm x 5mm x 5mm cube comes out (vernier caliper) comes out here with tolerances of 4.99mm to 5.02mm. Right next to it I am printing a cylinder with 5mm diameter. The cylinder has a diameter in the range of 4.74mm to 4.81mm. So a much larger discrepancy.

It looks like a somewhat constant offset. A 10mm cylindrical hole has an inner diameter of 9.80mm. Another 8mm hole has an inner diameter of approx. 7.8mm. 

This is with a 0.4mm nozzle in PLA using PrusaSlicer 2.5.0 with Arachne.

asteppke, 2022-11-26:
And while this issue was already discussed for classic mode (https://github.com/prusa3d/PrusaSlicer/issues/4561) the arachne mode makes it worse. So it seems even more urgent to provide a workaround or fix as a simple XY scaling of the complete model is not sufficient.

markmgit, 2022-12-31:
Cura has a feature called "HOLE Horizontal Expansion" to adjust for this. Maybe someone can do the same for Prusa Slicer?

mataksik, 2023-01-21:
Developers of Prusa Slicer insist on designin holes with this error in mind. They do not listen to needs of tolerances

asteppke, 2023-02-28:
And this issue becomes more prominent with the 2.6 feature of peg-like connectors which do not fit well together. There is one reason for this behavior discussed here: https://www.youtube.com/watch?v=yzPqBt2SrcE 

This slicer issue does not seem to present on competing FDM printers or there is at least a workaround in their slicer software. From the test data above it is also clear that this is not a hardware issue as rectangular features are spot on.

juanemartinboix, 2023-08-14:
I had read somewhere that the issue is related to any curve . When printing , the hot extruded line will eventually cool and retract more or less depending on materials CTE .In a straight line ,some tension will be created .In a rectangle the 4 lines will held each other .In a curve or a full circle ,the shrinkage is only supported by the previous layer and the result is a slightly smaller radius. The existence of a previous layer is apparently the reason for a not so proportional shrinkage .
If you slice a simple model like a cube with a hole with single walls ,no top/bottom ,nada ...is easy to see what is done to print the intended geometry.
If that is the confirmed case it will be trivial to add an offset to any generated arc based on some test ,even if the offset is size dependant.

Ruakij, 2024-02-18:
> If that is the confirmed case it will be trivial to add an offset to any generated arc based on some test ,even if the offset is size dependant.

SuperSlicer has an option called "Vertical Hole shrinking compensation" which does exactly that once it detected an inner enclosed circle-like structure as far as i know.
We would need to test how affected arcs are by themselves compared to full holes and if that needs further compensation.
I'd probably want a threshold for arc-angle to apply this.

Am pretty happy with SuperSlicer's customize-ability, but it is many versions behind and I am missing the new features of PrusaSlicer :/

rlneumiller, 2024-12-02:
Cura's hole compensation is a must have for small (sub 5mm) diameter holes.

MohrMa2, 2026-09-02:
Please do not close. This is not solved. 
Although I switched to orca due to this issue.

closed 2026-10-28 by PDBeal