prusa3d/PrusaSlicer #8710: Changing colors doesn't qualify as a change to save projct

MNeMoNiCuZ, 2022-08-18:
### Description of the bug

If you are editing a file in Prusa Slicer, and you have it saved as a .3mf-file, you are not allowed to save after you have made an edit to the color of a layer (right-clicking the layer in Preview to see what your color-change will look like).

This does not count as modifying the project, so the SAVE PROJECT-button is greyed out.

### Project file & How to reproduce

1. Add a model. 
2. Add colored layers in the preview mode (Tab to get to it after slicing).
3. Save project as a 3mf.
4. Change the color by right-clicking the color in the bar on the right side.
5. Click File-menu.
6. The Save Project-option is now grayed out.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0 beta 1

### Operating system

Win 11

### Printer model

MK3S

stepikovo, 2022-08-25:
Thank you, it's not only about the colour change but also adding a new command for another colour change is also not registered, we will check it.

YuSanka, 2022-08-25:
Thanks for your report, this bug will be fixed in 2.6.0