prusa3d/PrusaSlicer #8712: Slicer ALWAYS hangs when sending to multiple, sequential network printers.

oriddlero, 2022-08-18:
### Description of the bug

When you slice a model and send it to your first 3D printer it uploads fine. Now you change the target to the next 3D printer to print the same model, when you export the model does not upload.

### Project file & How to reproduce

1. Open a model and slicer
2. Send the print to printer number one over the network
3. Change the printer destination to another network printer with the same model open
4. Try to send it, it will not send.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2 

### Operating system

W10 

### Printer model

MK3+

kocikdav, 2022-08-23:
Hello, thank you for reporting. Unfortunately I'm not able to reproduce your issue. Would you share more information? How exactly is PrusaSlicer not sending? Does it freeze on 0 percent or shows some error message? Would you share a printscreen?

oriddlero, 2022-08-23:
It just freezes @ 0%, every time, the rest of the program is still responsive. No error message.  To get Prusa slicer to send the job to the 2nd printer you need to  save file>close>reopen file> reslice>send to 2nd printer.

kocikdav, 2022-08-24:
Thank you. We will look into it. Would you please confirm this issue is also on the latest release? https://github.com/prusa3d/PrusaSlicer/releases
We are close to releasing 2.5.0 final. So please test 2.5.0-rc1.

lukasmatena, 2023-02-06:
@oriddlero Could you please retest it with [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3)? Thanks.

closed 2026-09-29 by prusaslicer-bot[bot]