prusa3d/PrusaSlicer #8713: Rename NYLON filament type to PA

murk-sy, 2022-08-18:
Using NYLON as a general term for polyamides is a misnomer, as described in the PACF announcement blogpost:
[Introducing Prusament PA11 (Nylon) Carbon Fiber Black](https://blog.prusa3d.com/introducing-prusament-pa11-nylon-carbon-fiber-black-filament-with-excellent-chemical-and-heat-resistance_69135/)

> Polyamides are a group of polymeric materials produced since the 1930s when DuPont developed polyamide 6.6 known by its trade name Nylon. This name has since become widespread and is often incorrectly applied to other types of polyamides. They are referred to by numbers that represent the number of carbon atoms of the parent compounds – e.g. PA6, PA6.10, PA12, etc.

In my opinion trade names should not be used in filament types, although there are other examples like it currently in Prusaslicer. 

**Countering that**, there may be value in keeping it separate and adding other, different nylons, especially if there are major temperature/printing differences.
Additionally, some proprietary plastics do not have a (known) generic name.

rtyr, 2022-08-18:
You can use whatever filament type value you want, so I guess it should cover all needs.

[image]

murk-sy, 2022-08-18:
This change isn't really necessary by itself, it's more about consistency and changing it now, rather than deciding to do it when some future functionality relies on it.

Mini (and I imagine future printers too, don't know if MK3S has it right now?) now have a filament type check warning and will likely have more management features in connection with that, and that will almost assuredly be true for Prusa Connect as well.

The custom filament type isn't really "functional" right now on the Mini, since you just have to pick a material that's closest temperature wise anyway. For example, you have to use PC filament setting if you want to reliably use auto filament change with nylon (which is a standard type anyway). See also https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2416

rtyr, 2022-08-19:
That's true. Filament type in Prusa Connect and Buddy FW is certainly important aspect of this. We will discuss it internally.

rtyr, 2022-08-26:
Filament type NYLON is now changed to PA in all our system filament profiles (config update 1.5.1 for 2.5.0-alpha and newer). We will also add PA filament type to the selection (even though it is not needed). 

Closing.

closed 2022-08-26 by rtyr