prusa3d/PrusaSlicer #8715: [Enhancement] Separate cooling fan speed setting for overhangs

derenma, 2022-08-18:
**Is your feature request related to a problem? Please describe.**
Shallow overhangs that should not need supports could benefit from increased cooling.  Since drooping can still occur at angles slightly over a required support overhang angle, a separate Cooling setting would be nice to have in the Filament Settings tab.

Insufficient cooling while printing small overhangs can still lead to top layer warping and excessive drooping. Increasing the fan speed temporarily would be a nice alternative to adding additional supports. 

**Describe the solution you'd like**
Overhang Threshold Cooling in Filament Settings tab

**Describe how it would work**
Here are some conditions that I am thinking about:
1. If the Overhang Threshold is met in Print Settings, generate supports
2. If the actual overhang is within x% of the Overhang Threshold, increase cooling by y%

**Describe alternatives you've considered**
Increase part cooling for the entire duration of the print

rtyr, 2022-08-19:
Already requested in https://github.com/prusa3d/PrusaSlicer/issues/767. I am closing this as a duplicate, your issue will be linked.

closed 2022-08-19 by rtyr

derenma, 2022-08-22:
@rtyr 👍