prusa3d/PrusaSlicer #8716: [2.5.0-beta1] Arachne, 0.190 mm^3/s flow rate, actually leaving a hole in the part

Mithrandil, 2022-08-18:
### Description of the bug

in 2.5.0-beta1

The thicker perimeter is heavily underextruded, and therefore it leaves a gap in the print (verified on actual print)

In the picture you can see a line dark blue, printed at 0.19 mm^3/s, that's the problem, 

[image]

it DID NOT happen in alpha3 (below)

[image]

### Project file & How to reproduce

[BUG-2.5.0-beta1-Arachne 0.190 mm3s.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9377057/BUG-2.5.0-beta1-Arachne.0.190.mm3s.3mf.zip)

To reproduce just slice the attached file

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-beta1

### Operating system

Windows 10

### Printer model

Prusa MK2s with Klipper firmware

rtyr, 2022-08-19:
Thanks, we are investigating it. It seems to be caused by pressure equalizer.

bubnikv, 2022-08-19:
Fixed with upcoming PrusaSlicer 2.5.0-beta1

commit 8c0db761 (2022-08-19): Fix #8716, #8717, and #8718: The pressure equalizer was producing a malformed GCode that caused underextrusion.
closed 2022-08-19 by bubnikv