prusa3d/PrusaSlicer #8717: Very bad vertical wall quality in PrusaSlicer 2.5.0-beta1

sarusani, 2022-08-18:
### Description of the bug

When I slice with 2.5.0-beta1 I get very bad quality on some vertical walls. This does not happen when I use 2.5.0-alpha3 or older versions. I use the same settings in both versions. ([3mf](https://github.com/prusa3d/PrusaSlicer/files/9377938/outside_shell.zip)). 

I tested this with multiple materials to make sure it's not bad/wet filament and it is reproduceable with different models. (See pictures). I used the default 0.2mm quality profile (VERSION 1.4.7) with arachne enabled and pressure equalizer set to 3mm/s.

I do not know yet if this is caused by arachne or pressure equalizer, or something different. (Will update when I printed more tests)

This happens when I print from Octoprint & also from  SDCard.

PrusaSlicer 2.5.0-alpha3:
[image]
PrusaSlicer 2.5.0-beta1:
[image]

PrusaSlicer 2.5.0-alpha3:
[image]
PrusaSlicer 2.5.0-beta1:
[image]

### Project file & How to reproduce

[outside_shell.zip](https://github.com/prusa3d/PrusaSlicer/files/9377965/outside_shell.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-beta1+win64

### Operating system

Windows 10

### Printer model

i3 MK3S+ (3.11.1-RC1 Firmware)

rtyr, 2022-08-19:
Thanks, we are investigating it. It seems to be caused by pressure equalizer.

sarusani, 2022-08-19:
I turned off the pressure equalizer and everything is back to normal. Thanks for looking into it!

sarusani, 2022-08-19:
Sorry, pressed the wrong button... :)

bubnikv, 2022-08-19:
Fixed with upcoming PrusaSlicer 2.5.0-beta1

closed 2022-08-19 by bubnikv