prusa3d/PrusaSlicer #8718: Underextrusion caused by Pressure equalizer

Enanova, 2022-08-19:
### Description of the bug

Hello, with unchanged profile and settings it worked smoothly until now, but after an external update to the print settings 1-2 days ago, they broke and caused underextrusion on new prints. After turning off the Pressure equalizer in the slicer, the flow strength was fixed.

[image]
[image]
[image]
[image]

### Project file & How to reproduce

[image]
[image]
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0-beta1+win64-202208151053

### Operating system

Windows 11

### Printer model

creality cr 6 se

kubispe1, 2022-08-19:
Thanks, we are investigating it. Similar as:
#8716
#8716

bubnikv, 2022-08-19:
Fixed with upcoming PrusaSlicer 2.5.0-beta1

closed 2022-08-19 by bubnikv