prusa3d/PrusaSlicer #8719: 2.5.0 arachne Vase mode Issue

Design-Create-Innovate, 2022-08-19:
### Description of the bug

creating a flow test model using vase mode with Arachne type perimeter generation enabled produces odd separation of continuous printing path in one area of my model. disabling Arachne and reslicing with classic perimeter generation produces the model as expected with no issues.

### Project file & How to reproduce

[Max Flow Test.zip](https://github.com/prusa3d/PrusaSlicer/files/9382174/Max.Flow.Test.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0 alpha3 & Beta1

### Operating system

widows 10

### Printer model

Prusa mk3s

kubispe1, 2022-08-19:
Thank you for your report. I can confirm this issue in Prusa Slicer 2.5.0 beta1. Will be fixed in next release.
[image]

bubnikv, 2022-08-19:
Fixed with upcoming PrusaSlicer 2.5.0-beta1

closed 2022-08-19 by bubnikv