prusa3d/PrusaSlicer #8723: Graphics Error: System not running opengl

S1L3NTANG3L, 2022-08-20:
### Description of the bug

Get an error on startup regarding my system not running OpenGL 2.0 or higher, --sw-renderer works but found the origin of the bug -> GeForce Game Ready Driver 516.94.
After reverting back to 516.59 the error disappeared.
Thought I would just not it down here if anyone else get's this error.

### Project file & How to reproduce

Issue Driver -> https://www.nvidia.com/content/DriverDownload-March2009/confirmation.php?url=/Windows/516.94/516.94-notebook-win10-win11-64bit-international-dch-whql.exe&lang=us&type=geforcem
Supported Driver -> https://www.nvidia.com/en-us/drivers/results/190553/

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Windows 11

### Printer model

Ender 3 V2

kubispe1, 2022-08-22:
Hi, thanks for report. Can you provide us more info about your system (help tab / System info). Also I would like to ask you to try it in new release 2.5.0-rc1

S1L3NTANG3L, 2022-08-22:
I'll pull the version tonight and check, I'll also post the details here.

S1L3NTANG3L, 2022-08-22:
PrusaSlicer
Version:   2.4.2+win64
Build:     PrusaSlicer-2.4.2+win64-202204251110

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Edition	Windows 11 Pro
Version	22H2
Installed on	‎2022/‎05/‎13
OS build	22622.575
Experience	Windows Feature Experience Pack 1000.22634.1000.0

Total RAM size [MB]: 8,477MB
OpenGL installation
GL version:   4.6.0 NVIDIA 516.59
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1050/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 317 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

S1L3NTANG3L, 2022-08-22:
@kubispe1 Error does not appear to be showing up with 2.5.0-rc1 and GeForce Game Ready Driver 516.94.

closed 2022-08-22 by S1L3NTANG3L

kubispe1, 2022-08-22:
Thanks for your cooperation. Is help us.