prusa3d/PrusaSlicer #8724: Slicing when object almost fills the bed.

Bladevane, 2022-08-20:
### Description of the bug

Using Prusa Slicer 2.5.0-rc1. Tried to slice an STL which was 220 x 220. Slicer closed on me. Reduced size to 95% and sliced OK. Went up to 96% and it closed on me again.

### Project file & How to reproduce

[Weighing Machine Top.zip](https://github.com/prusa3d/PrusaSlicer/files/9387662/Weighing.Machine.Top.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-rc1

### Operating system

Windows 11 (Win64)

### Printer model

Prusa iMk3S

foreachthing, 2022-08-21:
Arachne causes the crash (on my computer).
If disabled (or set to classic), it slices fine.

Bladevane, 2022-08-21:
Thanks for the response. I tried it - and hey presto. Classic slices fine,
Arachne does not.

rtyr, 2022-08-22:
Reproduced. Thanks for reporting.

hejllukas, 2022-08-30:
Thank you for reporting this issue. It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/de48bcd1b2b8c479f2c2000258a9cf72c5f48829 in master_250, and it will be part of the next release.

closed 2022-08-30 by hejllukas