prusa3d/PrusaSlicer #8725: Asking to save file, PrusaSlicer 2.4.2

TheWebExpert1, 2022-08-20:
### Description of the bug

Every time I load a .stl file, when I quit the slicer, it's asking me to save the file.  This is even if no changes have been made.  This should be a thing which can be turned on and off.

### Project file & How to reproduce

This happens with every file.
[Untitled.zip](https://github.com/prusa3d/PrusaSlicer/files/9388000/Untitled.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+Win64

### Operating system

Windows 11

### Printer model

Prusa I3MkIIIS+

rtyr, 2022-08-23:
Go to the preferences and enable "Ask for unsaved changes in project"

[image]

That parameter is enabled by default, so you probably previously disabled it by checking "remember my choice" and anwering yes on this notification.

[image]

I am closing this issue.

closed 2022-08-23 by rtyr

TheWebExpert1, 2022-08-23:
That box is already unchecked, and it's still doing it.

reynol3232, 2023-11-25:
I realize this is old.  I have the same problem.  The box is unchecked and Prusa Slicer still asks to "Save Files As" when I start a new project.  I hate it.

rtyr, 2023-11-25:
It must be checked, not unchecked.