prusa3d/PrusaSlicer #8726: On single item with multiple parts, Z position has issues.

jgbrown54, 2022-08-21:
### Description of the bug

When loading multiple items as a single part. The single part can not be set for a Z of 0. If it is then it will show as sinking. Also parts do not start and end where specified. For example. One part is 1mm tall and begins at 0mm. The second through the 28th parts are .6mm tall and begins at .2mm. All parts should end at 1mm but they do not. The second through 28th parts seem to start at .6mm even though they are set with a Z of .4mm. To make it right, I have to set the start for those parts as .2mm. It seems like a zero origin issue.

### Project file & How to reproduce

[Stealie Coaster.zip](https://github.com/prusa3d/PrusaSlicer/files/9388227/Stealie.Coaster.zip)
Set parts 2 through 28 to have a Z of .4mm.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-rc1

### Operating system

Windows 11 X64

### Printer model

Longer3d LK5 Pro

kubispe1, 2022-08-22:
Hello, can I ask you for original .stl ?

jgbrown54, 2022-08-22:
I'll make a simpler one with about 3 or 4 parts. 2.5.0-rc1 was better than 2.5.0-beta but the issue was also in 2.4.2.

jgbrown54, 2022-08-22:
[ABTest.zip](https://github.com/prusa3d/PrusaSlicer/files/9393573/ABTest.zip)
The zip file contains 3 .stl files and a .3mf made from the .stl files.
The parts should be combined with the base starting at 0 and it is 1mm high. The Blue B and the Red A should each start at .4mm and they are .6mm high. So all 3 parts should end at 1mm.
The main part should start at 0mm and should be 1mm total in height. But if I try to set the Z to 0mm, it shows it as sinking.
With .2mm layer height, there should be 2 layers of solid black and 3 layers of black, blue, and red. But it comes out to 3 layers of black, 2 layers of black, blue, and red, and 1 layer of only blue and red with a total height of 1.2mm when it should only be 1mm maximum. To make it come out correctly, I have to set the Z for the blue and red to .2mm instead of .4mm.

kubispe1, 2022-08-22:
I think the problem is that all objects have to lay down on bed. If you import objects one by one (or together in selection) they always will pin to bed. And only parts of object can levitate.
Only solution I find, is to export assembly (for example from fusion) as a .step and then import it. This way respects object and its parts. Other formats like .stl or .obj joins parts together and make one object. But to be honest, maybe you can somehow achieve it also with .stl, but I don’t know. 
Here is my turn:
[assemble.zip](https://github.com/prusa3d/PrusaSlicer/files/9394003/assemble.zip)

jgbrown54, 2022-08-22:
[ABTest.zip](https://github.com/prusa3d/PrusaSlicer/files/9397846/ABTest.zip)
This zip file contains the STLs, a 3MF and a STEP. They all do exactly the same thing. When I import the STEP, all three parts come in but I can not adjust the Z for the combined part and the other parts do not combine correctly and can not be adjusted corectly. You say it won't work for SLT files then why did you ask me to send SLT files? If I import either the 3MF file or the 3 SLT files together, it asks me if it should be a single object with multiple parts so obviously it's meant to work. Or I can add the first part, then add the second and third part to it. No matter what, it always comes out the same. It also works in Slic3r which is where you started from. Why are you saying it shouldn't work with Prusa? I have the work around. Just offset all raised parts by .2mm. I just thought you might want to fix it.

jgbrown54, 2022-08-22:
I just noticed that you sent back a STEP file. It doesn't work either does it? The main object shows a Z of 6mm and if you try to set it to 0mm to place it on the bed, it shows as sinking. And component 2 starts at a Z of 1mm and goes 10mm high instead of starting at 0mm on the bed, Component 3 is stacked on top of component 2 but it has a Z of -5. If you think that makes sense then you should check your math.

jgbrown54, 2022-08-23:
[Prusa Pics.zip](https://github.com/prusa3d/PrusaSlicer/files/9398401/Prusa.Pics.zip)
Attached are snips of what I get when I open your "supposedly good" assemble.step.

Snip 1. This is what I get when I open assemble.step with no changes. Notice that assemble starts with a Z of 6mm instead of a Z of 0mm even though it's apparently on the bed.

Snip 2. If I set the Z position on assemble to 0, it shows as sinking.

Snip3. Component 2 shows as starting a Z of 1mm even though it is on the bed. It also shows as 10mm high.

Snip 4. Component 3 shows as starting at -5mm and it is 2mm high.

So let's add that up. If Component 2 is 10mm and Component 3 is stacked on top then that is 12mm. And assemble is 12mm high. So that's right. But if Component 3 starts at -5mm, and it's only 2mm high, it should never be seen because it's top is 3mm below the bed. And apparently Component 2 is floating above the bed. Not to mention that the whole assembly is marked as starting at 6mm above the bed. While things appear to be okay in the display, with the values shown it is unprintable. And if you look at the layer by layer you see that component 2 winds up with a hole in it the same size as component 3 but .4mm below where Component 3 begins.

And you think this is correct functionality?

kubispe1, 2022-08-23:
Fine, over this few messages, it is much more clear to me. Somehow I overlooked real meaning of issue, you are talking about.
In my previous assembly: 
z position of object is located in centroid of object box. So you see there 12/2 = 6 mm
z position of part is located given by centroid of position of object.  
So you see there for Component2 1 mm and for Component3 - 5mm
Hope more clear would be this picture:
[image]

Helpful could be also our help: 
https://help.prusa3d.com/article/how-to-lift-object-from-the-print-bed_245192

jb221, 2022-08-23:
Could we have an option to toggle z position between centroid and lowest point? I frequently have to manually align model parts and while I can compensate for this behavior it's definitely extra time spent doing so.

jgbrown54, 2022-08-23:
I'm glad you were able to clear that up. But I have to say it's very illogical. Wouldn't it make more sense to specify where a part starts and how high it goes like Slic3r?  referenced but it doesn't say anything that "centroid" garbage. At least now I know how to manipulate things correctly. This should be spelled out somewhere. Maybe a selection like the World and Local coordinates could be added for Centroid Positioning or Bottom reference.

jgbrown54, 2022-08-24:
I screwed around with this and finally figured out how this was working and there is a repeatable error. Follow closely.

Create a 1mm tall object. The Z wll be 1/2 the object height. If you add the same or another 1mm part to that object with a Right Click Add Part/Load, that part is added and you now have a multi-part object. Both parts have a Z of 0 and are 1mm tall. The multi-part object has a z of .5mm and is 1mm tall. Everything is good. If you take the second part and change it's Z to 1 and leave the size as 1. Everything is still good except now the multi-part item has a size of 2 which is right but the Z is still at .5mm and can not be changed. If you add another part and set it to a Z position of 2, things still appear correct and the parts are stacked as required and the Z size of the multi-part object remains at .5mm. The problem occurs when you change the Z size of one of the parts it tries to recalculate the z size for the multi-part object apparently based on the it's Z position. Now it is totally lost and even if you set the size of the part back to what it was. it only gets worse. That's why I was coming up with such crazy numbers. That didn't fit with what you were telling me. Even if I saved the project and reopened it, I couldn't get the numbers to match.

If all my parts had the correct size to begin with, I never would have had an issue. But because I had a multi-part object and changed the size any one of the objects, Nothing would make sense when adjusting the position after that.

jgbrown54, 2022-08-24:
[Test.zip](https://github.com/prusa3d/PrusaSlicer/files/9414149/Test.zip)
Let me simplify and explain how to recreat the error.
1. Open the 3 STL files simultaneously and answer Yes if it wants to know if it should be a single object with multiple parts.
2. Select each part in the property window and set the Z position on Part 1 to 0, set the Z position on Part 2 to 10, and set the Z position on Part 3 to 20.
3. You now have 3 blocks stacked on top of each other. This is what I was trying to say it should do. And it does. Part 1 starts at 0 and goes to 10. Part 2 starts at 10 and goes to 20. Part 3 starts at 20 and goes to 30. If you look the properties for the multi-part object, it has a Z position of 5 even though the total height is 30 and the centroid height is 15.
4. If you change the Z size on Part 3 from 10 to 20, the Z size of the multi-part object only changes to 35 even though the total height of the 3 parts is now 40. **That's` error 1**
5. If you change the Z size of Part 3 to 5, the Z size of the multi-part object changes to 27.5 even though the total height is now 25.
6. Setting the Z size on Part 3 to 10 will get you back where you started. The multi-part object has a Z size of 30.
7. If you set the Z position on Part 3 to 15 instead of 20, Part 3 is moved down and is partially embedded into Part 2. This is fine. And the Z size of the multi-part object has a Z size of 25 which is correct.
8. If you ,amually change the Z size of the multi-part object, the Z size is changed on all 3 objects proportionately. 
9. 9. Now even if you change the Z size of all 3 parts back to what they were, the Z size of the multi-part object does not go back to 30 like it was. **That's error 2**

Error 1: Occurs when you change the Z size of a part in a multi-part object. The Z size of the multi-part object is incorrect. If you attempt to change the Z size of the multi-part object, it will change the Z size of all 3 parts proportionaly.
Error 2: Occurs when you change the Z size of a multi-part object. The change is divided proportionaly among all of the objects. This may not be an error per se but it is very difficult to get back where you were. Undo can fix it but manual changes invariably fail,

Add these 2 errors together and you get to the point you want to through it out the window. If error 1 is fixed, then functionality is restored. And on error 2, don't change the Z size of a multi-part object.

kubispe1, 2022-08-25:
I also have found some older duplicate issues, where people struggled with z positioning of objects and modifiers. For better orientation i link it here and let issue open:
#8407
#3579

closed 2026-09-29 by prusaslicer-bot[bot]