prusa3d/PrusaSlicer #8727: 2.5.0-rc1 Cannot disable "Ensure Vertical Shell Thickness"

crashberrypi, 2022-08-21:
### Description of the bug

2.5.0-rc1 version
Slicer does not disable "Ensure Vertical Shell Thickness" When unchecked. Slicer still puts wiggle line next to inside walls that angle inward as it does when option is Checked.
I saw this was an issue for Version 2.3.3 also

### Project file & How to reproduce

[Bug_2.5.0-rc1.zip](https://github.com/prusa3d/PrusaSlicer/files/9388332/Bug_2.5.0-rc1.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-rc1

### Operating system

Windows 10

### Printer model

Anycubic Mega S

crashberrypi, 2022-08-21:
Forgot to say, the graphic preview shows that "Ensure Vertical Shell Thickness" is disabled, but the actual print and gcode still have the feature on.

stepikovo, 2022-08-29:
The "ensure vertical wall thickness" feature was implemented as an alternate implementation of the solid infill on sloping surfaces. When the check box is unchecked, the old algorithm from upstream Slic3r is used.
This will be reworked in the future.

As a workaround, you may use the layer modifier to set the top and bottom layers to zero.

Closing as a duplicate of #223, #6853 and many others.

closed 2022-08-29 by stepikovo

cimbalek, 2022-12-10:
Thank you for the hint with height range modifier! Works like a charm!