prusa3d/PrusaSlicer #8730: Printing in the air if the arachne engine used

ton4eff, 2022-08-21:
### Description of the bug

If Arachne is enabled, it prints in the air
[image]
.

### Project file & How to reproduce

If the classic option is enabled, there is no such a problem.
[SmartTemperatureTower_TempFloor_230.zip](https://github.com/prusa3d/PrusaSlicer/files/9388768/SmartTemperatureTower_TempFloor_230.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-rc1 win64

### Operating system

Windows 10

### Printer model

original i3 MK3S+

JohnOCFII, 2022-08-21:
Interesting.  As I use that model often, I tried and observed the same behavior in the preview (I did not actually print this today).  I used the default 2.5.0-rc1 settings for MK3S+ and Prusament PLA at 0.15 QUALITY settings.

al177, 2022-09-04:
I see this as well in 2.5.0-rc2 on the same model with default Prusament PLA configs, 0.6 nozzle MK3s profile, and 0.20mm DETAIL.

Decreasing "Minimum perimeter width" (min_bead_width) from the profile default of 85% to 50% removes the air printing on that structure entirely.

acockerham, 2022-10-01:
I am having this same issue with Arachne algorithm.  I have noticed it on a similar Temp Tower as well as a model that had been sliced with tree supports.  The last portion of the peaks do not slice, although the very tips do.

I tried to decrease Minimum Perimeter Width as mentioned by al177, but this did not change the slice.  Standard perimeter generator works fine.

Replicated on a brand new PrusaSlicer install. Build: PrusaSlicer-2.5.0+win64-202209060714

Print Settings = 0.20 mm NORMAL (0.4mm nozzle) @ CREALITY
Filament Settings = Generic PLA @CREALITY
Printer Settings = Creality Ender-3 Pro (0.4mm nozzle)

[image]

malibata, 2022-10-07:
It gets even worse if you check the print path and especially with thin bridges, for example with arachne engine enabled everything is printed in thin air and will fall
[image]
It prints perfectly with classic engine
[image]
Interestingly enough the Cura 5 has the same issues so it seems that it is a bug inherent to arachne engine...

malibata, 2022-10-07:
I found out that setting 'Perimeter transitioning threshold angle' (Under Advanced/Arachne...) to some very high value like 45 degrees fixes most of these issues...

Nailig, 2022-10-27:
I also have found bad slicing in arcane mode with several models when the print becomes small or thin. For me the problems are solves when i set Perimeter transitioning threshold angle to 40.

for example the latest model i tried:
https://www.printables.com/model/111129-prusament-filament-clip  -> the small support that needs to be cut out

Suffice to say that some tuning is needed.

progman32, 2023-01-03:
I'm printing a model with sharp teeth and I'm hitting this bug quite hard. Had to revert to classic perimeter generation, else almost every tooth ends up with pieces floating in midair. In case it helps, here's an 3mf of just one part of the tooth (I can't post the entire model, sorry).
[slicer test.zip](https://github.com/prusa3d/PrusaSlicer/files/10334622/slicer.test.zip)

[image]

Jan-Soustruznik, 2025-01-09:
closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7447

closed 2025-01-09 by Jan-Soustruznik