prusa3d/PrusaSlicer #8734: Arachne isn't always continuing external perimeters from internal perimeters when using large nozzles

iso250, 2022-08-21:
### Description of the bug

When an using a 0.6mm nozzle and slicing objects with T-intersection walls the slicer isn't continuing outer perimeters from inner perimeters.

You can see in the left cell it has pathed three lines and then moved on to another cell.

[image]

After doing the third line in the second cell it has returned to do the fourth line (and outer perimeter) in the first cell.

[image]

When printing with large nozzles at higher speeds gaps can be left when inner perimeters don't continue on to outer perimeters.  Additionally, the issue of inner perimeters not continuing on to outer perimeters was san issue that was supposed to be fixed many versions of PrusaSlicer ago.

### Project file & How to reproduce

[Paint Rack.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9389489/Paint.Rack.3mf.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-rc1

### Operating system

MacOS Monterey

### Printer model

Prusa Mk3s

iso250, 2022-08-22:
It seems to be most prevalent when there are T shaped walls, not single enclosed bodies.

iso250, 2022-09-16:
So disappointed this wasn't fixed.  Prusa makes a lot of noise about supporting large nozzles but when problems that affect large nozzle users happen we hear nothing and never see updates.

DragonFinlay, 2022-09-16:
We are addressing all issues, but we also need to assign different priorities to them.
Issues that crash the software need to be addressed before e.g. quality of life issues.

We have read your request and will try to fix it in the near future. Just give us some time please.

TerPuc, 2024-09-25:
Hello @iso250, I have stumbled upon your issue and noticed is it still open. We have made some changes in previous versions and I believe it is fixed, I am surprised the issue is still opened. Does it still persist or can I close this issue? 😸 Thank you for your feedback!

TerPuc, 2024-10-03:
Dear @iso250, since I have not received any reply from you I consider this issue solved and I close it. Thank you and a have nice day!

closed 2024-10-03 by TerPuc