prusa3d/PrusaSlicer #8735: Pressure equalizer should reduce pressure for tool changes, possibly other travel moves too

n8bot, 2022-08-21:
Pressure equalizer seems great, but it would be nice if tool changes were treated as zero volumetric rate, so that the very first and last moves of a tool start from zero volumetric rate.

Then, that leads me to other possibilities: reducing the volumetric rate for travel moves, depending on their length? Short travel moves might not need much reduction, but for a huge 200 mm travel move, it might be nice to slightly reduce the volumetric rate, then raise it again, after the long move.

Thank you for all the work on PrusaSlicer it is amazing.

DFEvans, 2022-08-21:
I wonder if your thought on travel moves would be the solution to the overextrusion seen on this print at the end of each printed line: https://i.redd.it/miatuqdl3vi91.png

The underextrusion at the start was improved by calibrating linear advance, but the overextrusion at the end has proved very difficult to combat.

bubnikv, 2022-08-22:
@n8bot your comment on zeroing volumetric pressure on tool change is valid. Regarding travel, we are considering to rework the pressure equalizer to take into account the simulated time / print speed using the firmware simulator we are currently using for print time estimates. This would allow us to recognize that the volumetric pressure drops significantly at direction change on infill lines for example. However such refactoring of pressure equalizer requires a lot of effort, thus it is not at the top of our list as of now.

n8bot, 2022-08-22:
Oh, BTW, @bubnikv, I had to convert all the `float` to `double` in the pressure equalizer code. Small layer height/extrusion width combos a precision error with the pressure equalizer -- the rate would fluctuate and jump around. It seems to be solved with double precision. I noticed that the actual X/Y/Z/E values emitted to G-code were being emitted with floats. So, not only was the rate fluctuating, but there were errors introduced in the amounts of plastic needed for each move, and perhaps errors in the X/Y location of the points but that is less likely.

closed 2026-09-29 by prusaslicer-bot[bot]