prusa3d/PrusaSlicer #8736: Disappearing control panels in 2.5 alpha and rc1

fast-tl, 2022-08-21:
### Description of the bug

When models are added to plater, the on-screen control panels (rotate, arrange, copy, paste, scale, etc) all disappear when the cursor is moved into the area of the screen where they can be interacted with.

### Project file & How to reproduce

1. add model or ideally multiple models, to plate
2. attempt to arrange them on the plate so that they are not all centered on the plate jumbled together.
[Camera_mount01.zip](https://github.com/prusa3d/PrusaSlicer/files/9389898/Camera_mount01.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0rc1

### Operating system

Windows 11Pro

### Printer model

Creality CR-10S Pro v1

kubispe1, 2022-08-22:
I cannot replicate your issue. (2.5.0 rc1, WIN 10 PRO)

bubnikv, 2022-08-22:
Did you ever had a chance to test an older PrusaSlicer version? What is
your computer configuration?

n8bot, 2022-08-22:
What graphics card does your computer have?

fast-tl, 2022-08-22:
It’s an integrated graphics card on a Dell Optiplex 790. Not many upgrade options.Sent from my iPhone

fast-tl, 2022-08-27:
I have video of the issue happening. How do I upload?Sent from my iPhone

stepikovo, 2022-08-29:
@n8bot please use any cloud storage and send a link then

lukasmatena, 2024-02-19:
No response, probably outdated, no other similar reports. Closing.

closed 2024-02-19 by lukasmatena