prusa3d/PrusaSlicer #8739: Monitor dimensions telemetry data incorrect.

msandy, 2022-08-22:
### Description of the bug

I was recently prompted to submit system information when I launched the app. When I clicked "view data verbatim" it showed mostly the same data that's present in Help > System Info, but it also includes information about monitors. On my 16" MacBook Pro M1 Max, it showed the following:

            "Monitors": [
                {
                    "width": "1728",
                    "height": "1117"
                }
            ]

In reality, the panel on this device is `3456 x 2234`, exactly twice the (per-dimension) size reported. This is almost certainly due to some kind of DPI adjustment, though I haven't applied any overrides to the app. I don't know what API is used to query screen size, but it should either be updated to reflect the true resolution, or expanded to include the per-dimension scale factor so it can be derived.

### Project file & How to reproduce

1. Launch app such that it triggers system info collection prompt.
2. Click "view verbatim data"

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.2+arm64-202204251109

### Operating system

macOS Version 12.5.1 (Build 21G83)

### Printer model

i3 MK3S+

FidelCapo, 2022-08-23:
I think the system info data contains the information that the system displays in the display settings, see screenshot.
[image]

msandy, 2022-08-31:
Exactly - my point is that this isn't likely the information you want in a hardware survey, which is generally used to determine what resolutions 2D assets should be sized as. The user's currently configured scale factor setting might be interesting, but I suspect it would be much more important to be able to distinguish whether a user has a 1080p or 4K display, even if the 4K user has scaling at 200% (resulting in a reported DIP size of 1080p).

closed 2026-09-29 by prusaslicer-bot[bot]