prusa3d/PrusaSlicer #8740: Archimedian Chords on top fill pattern skipping around and not filling large chunks

travislaynewilson, 2022-08-22:
### Description of the bug

I was printing a rectangular tray with a large, flat top layer.

I used the Archimedean Chords for my top fill pattern expecting what the Slicer app was showing me - a complete top layer with concentric circles starting from the middle outwards. 

Instead, the print started from the top-left of the print, filled in seemingly random bands at a time, and skipped large sections as it moved towards the center.

A quarter of the way through the top layer, it skipped to the center and began the center of the circles, completing beautiful, perfect circles until it was unable to create contiguous circles any further. 

It then proceeded to repeat the same initial process, creating fragmented bands and leaving out large sections until the layer was "complete".

### Project file & How to reproduce

[base.zip](https://github.com/prusa3d/PrusaSlicer/files/9396551/base.zip)

[image]

Simply print this file as-is.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Windows 10

### Printer model

Prusa MKS3+

travislaynewilson, 2022-08-22:
Here are some pictures of the finished product.

[image]

[image]

travislaynewilson, 2022-08-22:
A near-identical result happened with my next print using Octagram Spiral as my top fill pattern.

rtyr, 2022-08-23:
I believe this issue is basically a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5214. The top infill would have to be printed sort of monotonically, which is not so simple to implement for this infill type, see https://github.com/prusa3d/PrusaSlicer/issues/5214#issuecomment-755342741.

bobjackman, 2023-08-24:
Even if monotonic planning were too difficult, it would still be a huge improvement if it wouldn't jump around. It seems pretty random where some sections it'll print a couple lines, then skip ahead and do a few more, then come back and fill in between, then skip WAY ahead, then come back. It's all over the place, currently.

zjohnson16, 2025-01-13:
+1 Would love a monotonic Archimedean Chords

iamkim17, 2025-07-22:
Is there a fix for this one?

closed 2026-09-29 by prusaslicer-bot[bot]