prusa3d/PrusaSlicer #8742: [2.5.0-beta1] Arachne fails to slice a model

michael-kerbel, 2022-08-22:
### Description of the bug

When trying to slice the attached model it produces an error:
"Flow::spacing() produced negative spacing"
Only happens when increasing the perimeters to 3.

### Project file & How to reproduce

[IntegratedRewinder_v5.1_Wide_90mm-new-cap.zip](https://github.com/prusa3d/PrusaSlicer/files/9397568/IntegratedRewinder_v5.1_Wide_90mm-new-cap.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-beta1

### Operating system

Windows10

### Printer model

MK2.5S-MMU2

kubispe1, 2022-08-23:
Error confirmed for 2.5.0-rc1

melrhombus, 2022-08-26:
#8447 Was occurring in earlier releases as well

Currently getting it for some models regardless of perimeters
eg: https://www.thingiverse.com/thing:2878826

hejllukas, 2022-08-30:
Thank you for reporting this issue. Based on a report from testers, this was also fixed with https://github.com/prusa3d/PrusaSlicer/commit/614cd7081d4a33104272eb64d8f50939844ecbbd in master_250. This fix will be part of a next release.

closed 2022-08-30 by hejllukas