prusa3d/PrusaSlicer #8743: Can the Ender 3 S1 profile be fixed to include leveling in the start gcode? As the Ender3 Bltouch profile does?

avocatos, 2022-08-23:
**Is your feature request related to a problem? Please describe.**
The Ender 3 S1 has CR touch, and the Ender 3- BL touch profile has the correct gcode for leveling. It's a quick and easy fix on the ender 3 S1's profile start Gcode. It just has always sort of bothered me the ender 3 BL-touch has the correct gcode for the probe and leveling but the S1 profile does not.

**Describe the solution you'd like**
To correct the Ender 3 S1's profile to include the appropriate start Gcode for the S1's features. 

**Describe how it would work**
I'm quite new to gcode, however I do know there should be a code following g28 with anything that has CR touch/BL touch auto bed leveling.

I believe the code that should follow g28 autohome, is g29, but this would have to be double checked. Though it seems this is how the ender 3-Bltouch profile goes

**Describe alternatives you've considered**
Having to add the code my self, and having to let others know that the profile does not default include the leveling start g-code like the ender-3 BLtouch profile does.

**Additional context**
It's a quick and easy fix, so is doing it your self, however I can see this being beneficial for newer people to printing to get their first print going, as they go on to learn gcode.

cnlson, 2022-08-23:
needs to be moved to a feature request, it is  not a bug

rtyr, 2022-08-23:
@pmjdebruijn what do you think

pmjdebruijn, 2022-08-23:
@avocatos thanks for reporting!
@rtyr: see #8746

rtyr, 2022-08-24:
Fixed in 0.2.1 configuration update. Closing.

closed 2022-08-24 by rtyr

avocatos, 2022-08-24:
> Fixed in 0.2.1 configuration update. Closing.

Hi!
How could I receive this update? Do I reinstall prusa slicer? 

Thanks

rtyr, 2022-08-24:
Just start the PrusaSlicer at least twice, then you should get notification update. The update is available only for PrusaSlicer 2.5.0-alpha and newer, the latest 2.5.0 release is rc1, available for download here https://github.com/prusa3d/PrusaSlicer/releases.

pmjdebruijn, 2022-08-24:
@avocatos would please report back if the new configuration update results in the proper behavior?

pmjdebruijn, 2022-09-15:
@avocatos you never reported back, whether everything worked well for you?

I'd also like to know which firmware version you're testing this on...?