prusa3d/PrusaSlicer #8744: random floating bits in 2.5 rc1

cnlson, 2022-08-23:
### Description of the bug

when slicing the same model in 2.4.2 and 2.5 rc1 random floating pieces show up in the same model when sliced with 2.5

I was looking for perhaps more resolution in 2.5 under arachne than with the traditional, so i used the  The_Golden_Age_of_Chivalry_circus_wagon_kit downloaded from thingiverse. it has features that i am not able to normally resolve with a 0.4mm profile so i was wondering what would happen with 2.5 and what happened would not work well. 

STL https://www.thingiverse.com/thing:639901

3mf from 2.4.2
[2.4.2_dragonfloatrwithdriver.zip](https://github.com/prusa3d/PrusaSlicer/files/9399286/2.4.2_dragonfloatrwithdriver.zip)
image 
[image]

3mf from 2.5 rc1 
[2.5rc1_dragonfloatrwithdriver.zip](https://github.com/prusa3d/PrusaSlicer/files/9399291/2.5rc1_dragonfloatrwithdriver.zip)
image 
[image]

as you can see in either the 3mf or the image, there are 2 spots where printing is randomly out in the air without any attachment to anything.  

### Project file & How to reproduce

uh project files are included in the description. but here they are again 

3mf
2.4.2
[2.4.2_dragonfloatrwithdriver.zip](https://github.com/prusa3d/PrusaSlicer/files/9399301/2.4.2_dragonfloatrwithdriver.zip)
2.5 rc1
[2.5rc1_dragonfloatrwithdriver.zip](https://github.com/prusa3d/PrusaSlicer/files/9399302/2.5rc1_dragonfloatrwithdriver.zip)

images
2.4.2
[image]

2.5 rc1

[image]
 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5 rc1

### Operating system

windows 10 x64

### Printer model

prusa mk3s+

kubispe1, 2022-08-23:
Same issue as
#8730

cnlson, 2022-08-23:
@kubispe1 thanks for that, I searched but obviously not with the correct terms.

Ok to close as dup