prusa3d/PrusaSlicer #8745: multimaterial wrong slicing on edge

TylonHH, 2022-08-23:
### Description of the bug

Hi,
I have here an architecture modell wich I wanna print in multicolour. So after spending hours colouring all the facetten I have one edge wich is wrong sliced. I tried to rotate the modell, but the bug still persist.
Then I thought I could cut the edge to get different triangle, no luck.

So for me it seems to be a bug.
What you think?

PS: I also posted in the [community forum](https://forum.prusa3d.com/forum/prusaslicer/wrong-slicing-with-multicolour/#post-618036).

### Project file & How to reproduce

Open project and slice. Dont worry about the bed size, I hoped to see artefacts at the corner.
[Model1_Final_WIP.3mf.TXT](https://github.com/prusa3d/PrusaSlicer/files/9399972/Model1_Final_WIP.3mf.TXT)

[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.2+win64-202204251110

### Operating system

Win10

### Printer model

Prusa

kubispe1, 2022-08-31:
This issue persist in 2.5.0 alpha 3 and also 2.4.2 tested on Win 10.

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena