prusa3d/PrusaSlicer #8747: 2.5 RC1 crash when slicing

tjhawkey, 2022-08-23:
### Description of the bug

Crashes and shuts down when slicing this project
[Spool-holder-R.zip](https://github.com/prusa3d/PrusaSlicer/files/9411036/Spool-holder-R.zip)

### Project file & How to reproduce

Click "slice now" button

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5 rc1

### Operating system

Windows 10

### Printer model

MK3S+

kubispe1, 2022-08-24:
Hi, thanks for your report and attached model.
I checked it little bit and there are some points, I discovered.:
- When you repair model in Netfabb:  It works but it is sinking
- According original settings the slicing in 2.5.0-rc1, 2.5.0-beta1, 2.5.0-alpha3: Does not work
- When you reset or edit variable layer height (Arachne is on):	 slicing is OK
- When you let layer variable layer height and switch Perimeter generator (from Arachne) to Classic: slicing is OK

My conclusion: 
The problem could be between variable height and Arachne generator for this model…
? Or Could be problem with a model?

lukasmatena, 2022-08-29:
The crash is likely related to #8761.

hejllukas, 2022-08-30:
Thank you for reporting this issue. It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/de48bcd1b2b8c479f2c2000258a9cf72c5f48829 in master_250, and it will be part of the next release.

closed 2022-08-30 by hejllukas